espruino/EspruinoDocs #544: Bangle.js tutorial wishlist

gfwilliams, 2020-03-27:
This is a list I'll use to prioritise which tutorials get written first.

Please vote on each entry here with the 'smiley face' icon, thumbs up or thumbs down.

If you want to add a suggestion, please add one suggestion per reply so they can be voted on.

gfwilliams, 2020-03-27:
Creating Widgets

gfwilliams, 2020-03-27:
- [ ] Collecting/saving data to Bangle.js and creating an 'interface.html' file to read it onto a PC

gfwilliams, 2020-03-27:
- [ ] Adding `custom.html` to create a customisable app (eg QR Code)

gfwilliams, 2020-03-27:
- [ ] Using Gestures (Tensorflow)

gfwilliams, 2020-03-27:
- [ ] Controlling other Espruino devices via Bluetooth

gfwilliams, 2020-03-27:
- [ ] Using Bluetooth Scanning to output data broadcast by other devices

gfwilliams, 2020-03-27:
- [ ] Opening Bangle.js and adding a Piezo Speaker

gfwilliams, 2020-03-27:
- [ ] Opening Bangle.js and connecting external hardware

gfwilliams, 2020-03-27:
- [ ] Using GPS for positioning

kikura, 2020-03-28:
Difference between normal apps and watch faces.

MaBecker, 2020-03-28:
- [ ] secure UART connection via whitelist

jumjum123, 2020-03-28:
Using Gestures (Tensorflow)

MaBecker, 2020-03-30:
- [ ] How to write multi language apps/widgets.

neshanjo, 2020-04-05:
- [ ] setting up Visual Studio Code for development and deployment (e.g. via npm scripts)
Should also work with Windows 10.

unmotivagedgene, 2020-04-11:
Communicating to/from Bangle JS app/widget to/from native Android App.

gfwilliams, 2020-04-17:
- [ ] [Using DroidScript](http://droidscript.org/) to communicate with Bangle.js from Android

4thex, 2020-06-27:
I have not received my Bangle.js yet, but one of my first ideas is to read the accelerometer and the magnetometer together to accurately calculate position. Any hints on the calculations would be appreciated.

ClintH, 2021-11-17:
GET/POST to a HTTP JSON endpoint via the forthcoming 'bridge' app?

cyberchucktx, 2021-11-23:
Document the IOS app : how to read/cancel/remove notifications on the watch.

cyberchucktx, 2021-11-23:
Using MQTT client on the watch to be used with an MQTT Bluetooth bridge such as a phone/tablet (iPHone in my case) or an MQTT bridge (separate hardware such as this GL-S10 bridge @ [https://store.gl-inet.com/products/gl-s10-bluetooth-iot-gateway?variant=39514685145182](https://store.gl-inet.com/products/gl-s10-bluetooth-iot-gateway?variant=39514685145182)

cyberchucktx, 2021-11-23:
Using IFTTT on the Bangle2 watch  (?possible?)

github-oa, 2021-12-09:
"Remote Control" for the camera app on the smartphone. A simple "release button" for selfies, so we can take pictures from a distance would even do.

Razoxane, 2021-12-13:
How to add a settings menu to a widget or app, with all the various ways of presenting the options.
I have been trying to add a multiple choice picker, and have had to resort to copying other apps implementations, with varying levels of success.

gfwilliams, 2021-12-13:
@Razoxane have you seen https://www.espruino.com/Bangle.js+App+Settings ?

Razoxane, 2021-12-14:
> @Razoxane have you seen https://www.espruino.com/Bangle.js+App+Settings ?
> 
> But is the issue more that it should cover how to do multiple choice?

More how to do multiple choice, and also where that link is included in the docs.
That link doesn't appear in the https://www.espruino.com/Tutorials page, which was where I'd first checked. 

Now that I know of its existence, after going hunting to find what information I'd come across that was inadequate, I can see that https://www.espruino.com/Bangle.js#tutorials has a link to the Settings tutorial.
I can also see that the site search (rather than browser search) returns the Settings page.

gfwilliams, 2021-12-14:
Ahh - thanks. Looks like I hadn't updated the Tutorials page - I'll do that. I'd definitely recommend looking on https://www.espruino.com/Bangle.js2#tutorials  - as those should be tailored specifically for the watch

splch, 2022-09-08:
Is there a tutorial on how to add to the Bangle library? I just made a PR but didn't know how to test the code I wrote. That sort of development tutorial would be very useful :)

gfwilliams, 2022-09-08:
@splch Ahh - you mean extending Espruino itself? eg your PR https://github.com/espruino/Espruino/pull/2264 ?  there is something that may help you (http://www.espruino.com/Extending+Espruino+1), but in this case I think making a simple JS library will be easiest. I'll explain in the PR and will add an example.