espruino/EspruinoDocs #547: Swipe between (pre-selected) apps that run in the background.

psbest, 2020-04-02:


gfwilliams, 2020-04-02:
More information? is this about Bangle.js? If so it should be in the BangleApps repo.

gfwilliams, 2020-04-02:
Ahh - did you mean to add this to: https://github.com/espruino/EspruinoDocs/issues/544 ?

psbest, 2020-04-02:
yes, that is fine, thank you.

gfwilliams, 2020-04-04:
I don't really understand what you mean? Is this more of a feature request?

Apps don't run in the background, so I don't really know how I'd do a tutorial on it...

psbest, 2020-04-04:
I am looking for a way to have something like tiles (screens) that you can
swipe, for example time, navigation, health, timers without fiorst going to
the menu. If they can' t run in the background a way to switch between
them, like the widgets then without a active screen, by collecting data so
when you swipe to the tile you have instant info up-to-date, So it comes to
a way to distribute the cpu time betweens tiles the best way. How can i set
this up.