espruino/EspruinoDocs #548: info/Troubleshooting Bangle.js.md

paultanner, 2020-04-04:
Clock did not switch to summer time.  Have set it manually but not sure why it could not pick it up from phone?

gfwilliams, 2020-04-04:
Did you use `Set Current Time` on the App Loader?

If you're using Gadgetbridge, this was fixed (https://github.com/espruino/BangleApps/issues/170) and will be in the next Gadgetbridge update

paultanner, 2020-04-04:
No I set it in Settings > Set time.  Is that wrong?

gfwilliams, 2020-04-04:
> `Settings > Set time`

On the phone?

Not sure I understand - so you didn't connect it to your phone in any way, but you're surprised that the phone didn't manage to send it daylight saving time information despite never having a connection?

There is no daylight saving time built into the watch itself. There's a discussion on the forum about this: http://forum.espruino.com/conversations/345273

closed 2020-09-21 by gfwilliams