espruino/EspruinoDocs #550: tutorials/Bangle.js First App.md, Upload icon image

neshanjo, 2020-04-05:
Hi,
I followed the tutorial to build my first app. Now I'd like to set an icon. However, I cannot find the "Storage" icon.
```
Now it's time to convert the image for use in Espruino and upload it. You can use
http://www.espruino.com/Image+Converter for this, but the tools are now built 
into the IDE.

- Click the Storage icon above Upload: [ICON]
- Choose Upload a File
```
Am I just overseeing it? Or has this funcionality been removed from the Web IDE?

Thanks in advance!
Johannes

neshanjo, 2020-04-05:
I'm now seeing it 😕 
Sorry for this "spam"...

closed 2020-04-05 by neshanjo

neshanjo, 2020-04-05:
I now know why I was confused: The "Store" icon is not shown in the chrome app. Must be an older version.

gfwilliams, 2020-04-06:
Ahh, thanks! Yes, I need to update the Chrome app. I've been a bit slow with it lately as Google is supposed to be getting rid of them extremely soon