espruino/EspruinoDocs #566: devices/arduino-motorshield.md

Isanderthul, 2020-08-09:
The code does not seem to work, in addition the pin assignment in http://www.espruino.com/modules/arduino-motorshield.js is duplicated, 
```
    LATCH:arduino.D12,
  ...
    PWM2A:arduino.D12, // L293D 2
```
This should be `PWM2A:arduino.D11`. 
Seems to me this code never worked, and was never tested. Is there anyone who could help getting this board working?

gfwilliams, 2020-08-17:
What device are you using this on, Pixl.js?

I have used this board in the past, but it's possible I didn't check all the outputs worked...

Isanderthul, 2020-08-17:
I am using this on a pixl.js, I am using a banggood board. I made a few updates to the code and module, I am not sure if you are happy reviewing the proposed updates https://github.com/espruino/EspruinoDocs/pull/569.

Isanderthul, 2020-08-17:
I dont fully understand how modules are maintained, so if there is a potential for conflict, perhaps I should just put this in the forums.

gfwilliams, 2020-08-18:
Thanks for that - it looks good.

So what was the underlying issue? It seems that PWM2A wasn't working, but was your main issue just to do with powering the shield as you'd mentioned in https://github.com/espruino/EspruinoDocs/pull/569/files#diff-17fdeaa63329d03df0b8e36079c538abR34?

commit 3ae941a4 (2020-11-05): Tidy up - remove warnings of a short. It's not the case at all (tested here and fine) - fix #566
closed 2020-11-05 by gfwilliams