espruino/EspruinoDocs #570: UI efficiency

d3nd3, 2020-08-19:
Top right of the docs is a search field, its super useful.  Except I have an issue with it not hiding its search results after i select what i wanted in the list, requiring me to put an extra action for every search (highlighting search-text and deleting it).  

Would be nice if you could fix that.

d3nd3, 2020-08-19:
`--- search.js	2020-08-19 08:07:01.082552000 +0100
+++ search.js	2020-08-19 08:07:05.312160600 +0100
@@ -144,6 +144,9 @@
         html = '<div class="panel panel-default"><div class="panel-body">'+html+'</div></div>';
 
         $("#searchpopup").html(html).offset(pos);//.show().blur(function() { $("#searchpopup").hide(); });
+        $("#searchpopup a").click(function() {
+          $("#searchpopup").hide();
+        });
       } else {
         $("#searchpopup").remove();
       }
@@ -165,12 +168,29 @@
    }
 
 $(function() {
-   $(".searchbox").on('input', function(){doSearch($(this));}).keypress(function(event) {
+  $(".searchbox").click(function(e) { 
+    e.stopPropagation();
+    $("#searchpopup").show();
+    doSearch($(this));
+  }).on('input', function(){
+    $("#searchpopup").show();
+    doSearch($(this));
+  }).keypress(function(event) {
     if (event.keyCode == 13) {
       event.preventDefault();
-      if (urls.length == 1)
+      if (urls.length == 1)  {
         window.location = urls[0];
+        $("searchpopup").hide();
+      }
+    }
+   }).keyup(function(event) {
+    if ( event.key == "Escape" ) {
+      event.preventDefault();
+      $("#searchpopup").hide();
     }
    });
+  $(document).click(function() {
+    $("#searchpopup").hide();
+  });
 });
 var resultNames = [];
`

gfwilliams, 2020-08-19:
That's great - thanks! Usually when you click the search box it'll reload a new page. I hadn't considered the case where you're already on the reference and are going to another reference page :)

gfwilliams, 2020-08-19:
Thanks! I just pulled this in - it'll go live the next time I push a website update

closed 2020-08-19 by gfwilliams