espruino/EspruinoDocs #575: tutorials/Bangle.js Custom.md

computermacgyver, 2020-08-29:
The JavaScript file customize.js needed for the Bangle.js App Customiser is located at `core/lib/customize.js` within the [BangleApps(https://github.com/espruino/BangleApps) repo, but on http://banglejs.com/apps the the example script is at https://banglejs.com/apps/lib/customize.js .

As a result, apps using the 'customiser' approach do not work with GitHub pages. I.e., all customizer apps are broken at:
https://espruino.github.io/BangleApps/
and the tutorial here which includes:
>    <script src="../../lib/customize.js"></script>
does not work 'out-of-the-box'

It didn't take me long to figure this out, but I worry whether it might be a roadblock for others. I can see two solutions and wonder which @gfwilliams would prefer. 
1) Put the `lib` folder in the root of the BangleApps GitHub repo (easy for developers, but messy file structure and requires changes elsewhere I'm sure)
2) Update `tutorials/Bangle.js Custom.md` to describe a workaround (I would suggest adding a symlink and running a local webserver --- only update required is to the tutorial)

I'm happy to have a go at updating the tutorial if you think 2 is the better approach, Gordon. Thanks!

gfwilliams, 2020-08-31:
Hi - Sorry about this. There was a recent refactor (last week) which moved things around - and banglejs.com/apps and the espruino website haven't yet been updated.

I think the docs on GitHub are updated with the new URL (https://github.com/espruino/EspruinoDocs/blob/master/tutorials/Bangle.js%20Custom.md) but the copy on Espruino.com isn't yet - i'll do that in the next hour.

>  I.e., all customizer apps are broken at: https://espruino.github.io/BangleApps/

Are you sure about this? I just tried and it worked fine.

Basically I did the refactor to allow me to use the same code for a more generic Espruino device app loader, but I didn't want to push to banglejs.com/apps immediately unless something was broken.

But I'm pretty sure it's ok. If something isn't working for you, please can you post up which app and what you did?

computermacgyver, 2020-08-31:
Thanks, Gordon. That's great.

I've just checked again, and the only customizer app that is broken is mine (World Clock) because I put in the old paths matching what I saw on http://banglejs.com/apps . I'll fix that and send a PR momentarily.

closed 2020-08-31 by computermacgyver

gfwilliams, 2020-08-31:
Thanks! And sorry - I thought I'd picked up all the customisers and fixed them. Must have just been bad timing :(

computermacgyver, 2020-08-31:
No worries. I just missed it and followed the old pattern :-)