espruino/EspruinoDocs #577: devices/ADS1X15.md

edgarbjorntvedt, 2020-09-19:
channel 0 and 3 is working as expected, but 
channel 1 is mirroring channel 0 and 
channel 2 is mirroring channel 3

To verify that it is a software issue in Espruino/SX127x.js, I have created an identical version of the code in Arduino, which is working as expected and according how I moved the analog potentiometers. Therefore the Arduino version can be considered a good reference.

Im using NodeMCU/ESP8266, with ADS1115. My hardware setup is almost following the schema in the image, except that ADDR is connected to D3 on NodeMCU which I have set to LOW in code. I've also connected 4 potentiometers to the 4 channels:
[image]

When running the two tests, I tried to move the potentiometers identical. One test run for Arduino version and one for Espruino version. The outputs is showing the error described above. 

Please let me know if I've missing something in my test setup.

```
// Espruino version
var i2c = new I2C();
i2c.setup({sda:NodeMCU.D2, scl:NodeMCU.D1});
digitalWrite(new Pin(NodeMCU.D3), 0);

var ads = require("ADS1X15").connect(i2c);
ads.setGain(6144); // +/- 0.256mV

var v1, v2,v3,v4;

console.log('Espruino version');

setInterval(function() {
    ads.getADC(0, function(val) {
        v1 = val;
    });
    ads.getADC(1, function(val) {
        v2 = val;
    });
    ads.getADC(2, function(val) {
        v3 = val;
    });
    ads.getADC(3, function(val) {
        v4 = val;
    });
    console.log(v1, v2,v3,v4);
}, 500);

// output:
>Espruino version
undefined undefined undefined undefined
13228 13228 2767 2767
13229 13229 2769 2769
13229 13229 2769 2769
13229 13229 2769 2769
13229 13229 2769 2769
13225 13225 7127 7127
13268 13268 24617 24617
4836 4836 24647 24647
1985 1985 24556 24556
1984 1984 24543 24543
1987 1987 24556 24556
1986 1986 24554 24554
13235 13235 24638 24638
13235 13235 24630 24630
13235 13235 24630 24630
13236 13236 24630 24630
13236 13236 24629 24629
13235 13235 24619 24619
13235 13235 24619 24619
13235 13235 24630 24630
13236 13236 24629 24629
13235 13235 24630 24630
13235 13235 24630 24630
13236 13236 24631 24631
```

```
// Arduino version
#include <Wire.h>
#include <Adafruit_ADS1015.h>

Adafruit_ADS1115 ads;  /* Use this for the 16-bit version */

void setup(void) 
{
  pinMode(D3, OUTPUT);
  digitalWrite(D3, LOW);
  
  Serial.begin(9600);
  delay(500);
  
  Serial.println("Arduino version");
  ads.setGain(GAIN_TWOTHIRDS);  // 2/3x gain +/- 6.144V  1 bit = 3mV      0.1875mV (default)
  ads.begin();
}

void loop(void) 
{
  int16_t adc0, adc1, adc2, adc3;

  adc0 = ads.readADC_SingleEnded(0);
  adc1 = ads.readADC_SingleEnded(1);
  adc2 = ads.readADC_SingleEnded(2);
  adc3 = ads.readADC_SingleEnded(3);
  Serial.print(""); Serial.print(adc0);
  Serial.print(" "); Serial.print(adc1);
  Serial.print(" "); Serial.print(adc2);
  Serial.print(" "); Serial.println(adc3);
  
  delay(500);
}

// output
Arduino version
13239 14871 14145 2768
13239 14879 14144 2768
13239 14878 14144 4249
13256 14878 14142 24107
13254 14879 14145 24545
4355 14879 14145 24581
1951 14879 14148 24535
1954 14879 14142 24534
13235 14878 14147 24606
13232 14878 14147 24601
13232 14879 14146 24585
13232 14873 14145 24594
13232 14878 14145 24595
13232 14878 14145 24596
13231 14878 14144 24597
```

edgarbjorntvedt, 2020-09-19:
Debugged the i2c communication and found the error in my own code. The difference between the apis is that the Espruino api is an async api. The Arduino api is synchronous. So here is the updated working code for Espruino:

``` javascript
var i2c = new I2C();
i2c.setup({sda:NodeMCU.D2, scl:NodeMCU.D1});
digitalWrite(new Pin(NodeMCU.D3), 0);

var ads = new ADS1X15(i2c);
ads.setGain(6144); // +/- 0.256mV

var v1, v2,v3,v4;

console.log('Espruino version');

setInterval(function() {
    ads.getADC(0, function(val) {
        v1 = val;
        ads.getADC(1, function(val) {
            v2 = val;
            ads.getADC(2, function(val) {
                v3 = val;
                ads.getADC(3, function(val) {
                    v4 = val;
                    console.log(v1, v2,v3,v4);
                });
            });
        });
    });
}, 500);
```

closed 2020-09-19 by edgarbjorntvedt

gfwilliams, 2020-09-21:
Nice - thanks for the update and closing the issue :)