espruino/EspruinoDocs #581: Non-working example: UI Example - all ui elements on Pixl.js

cmumford, 2020-10-28:
https://www.espruino.com/uiExampleAllPixl doesn't display correct JavaScript. Here is one example:

```javascript
// ---------- Slider examples ----------
var s2 = // hold on to "s2" slider ui element for direct access app later on.
//....0  1     2       3   4   5   6   7   8  9       10  11                         12
//.flgs  clazz id      x   y   w   h  bc fcs  value   cb, ls (arr of labels (.],..]))
//       sli                ->x2->y2                    [0] fs tc    x  y  txt // formatter
ui.c( 3,"sli","s2"  , 12, 46,102, 16,  0,[0,7],  33,  ca, , u,  40, 7,   u
```

Note the missing parameter after `ca,`.  Alternatively if the raw code is copied from https://github.com/espruino/EspruinoDocs/blob/master/examples/uiExampleAllPixl.js into the WebIDE, without the last line containing setTimeout(), then the following error is encountered during startup:

```
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v08 (c) 2019 G.Williams
>
>save()
=undefined
Compacting Flash...
Calculating Size...
Writing..
Compressed 40000 bytes to 22083
BLE Connected, queueing BLE restart for later
Running onInit()...
Uncaught ReferenceError: "l" is not defined
 at line 2 col 157
...*k,d*k,e,c,b*h,d*h,b*n,d*n,l,p]}return b
                              ^
in function "sliM" called from line 1 col 41
...;m=b.sliM(m,[c,d,g,e],n,a,f);l=m[6+f];p=m[8+f];var r;q&&b.sl...
                              ^
in function "sliDu" called from line 1 col 75
...,c,d,g,e,k,h,n,void 0,p,m,1);0<--b;)q.ld(p[b],
                              ^
in function called from line 1 col 30
b[0]&1&&a[b[1]+"D"].apply(a,b)
                             ^
in function called from line 1 col 78
...&&a[b[1]+"D"].apply(a,b)},a);c&&c(a,b);return a
                              ^
in function "d" called from line 144 col 15
      .d(u,xDs)             // display all elts incl. extra ...
              ^
in function called from system
> 
```

gfwilliams, 2020-10-28:
Wow, ok - thanks! So the webpage rendering looks like it was broken (on this, and one or two other pages). I have now fixed that.

However I'm not sure about the other errors. I wonder whether they are due to some minification errors as with the IDE set up to use unminified modules it all seems ok

cmumford, 2020-10-28:
@gfwilliams thx for the webpage fix. If I understand correctly you are saying that

cmumford, 2020-10-28:
Oops - closed by accident.

gfwilliams, 2020-11-05:
Ok, I just tweaked the modules so that these ones in particular only do whitespace minification (as they are pretty dense anyway) and it all works fine now

closed 2020-11-05 by gfwilliams