espruino/EspruinoDocs #584: uiExampleAllPixl cannot load using "Send to Web IDE" button

cmumford, 2020-11-09:
The "Send to Web IDE" button at https://www.espruino.com/uiExampleAllPixl consistently fails with the following error:

```
Request-URI Too Long
The requested URL's length exceeds the capacity limit for this server.
Apache/2.4.18 (Ubuntu) Server at banglejs.com Port 80
```

The URL is 13,895 bytes. Is it possible to switch to HTTP POST to transport the code to the Web IDE? If I manually transport the code via the clipboard then the sample app works.

gfwilliams, 2020-11-09:
Ahh, ok - it may be the most sensible option right now is to just remove the button for requests that are too long. As mentioned in the other bug you filed, `uiExampleAllPixl` is a particularly dense example - the vast majority of code examples are significantly smaller and there is no issues with uploads.

As it happens, the query string isn't even interpreted by the server - it's actually the Web Browser that grabs the code from the query string, so it doesn't matter if the server can't handle it. I guess it could be possible to use local storage/similar to handle it.

cmumford, 2020-11-09:
Agreed. The next biggest one is uiExampleSli.js at 12440 KB. I'm not sure what device that runs on, as there is a uiExampleSliPixl.js file.

gfwilliams, 2020-11-10:
Actually the odd thing is that I just tried and for me, https://www.espruino.com/uiExampleAllPixl works great - because with the Chrome app installed it's picked up and never touches the server.

I've just added code to remove the button for big code samples though. It'll go live soon, so closing this one

closed 2020-11-10 by gfwilliams