espruino/EspruinoDocs #585: All UI elements seem to fail on uiExampleAllPixl.js

cmumford, 2020-11-09:
The UI elements in uiExampleAllPixl.js seem to fail in a variety of ways on my 2v08 PIxl.js device. If I use the lower-left button to navigate focus to an item, and then press the lower-right button I get the following errors, depending on the control:

    Execution Interrupted during event processing.
    New interpreter error: MEMORY

or:

    id: r1.c val: S untouch focused
    Execution Interrupted during event processing.

or: 

    id: b3 val: 3 untouch focused
    c: sli c: sli id: s2 v:  f: 0000 1000 0110 undefined
    Execution Interrupted during event processing.

However, I can control the backlight.

gfwilliams, 2020-11-09:
Did you upload to RAM, or to Flash?

`uiExampleAllPixl.js` is a bit of a brutal test on the Pixl - loading everything in at once.

Chances are because of the complexity of the UI framework, loading the whole thing and the example into RAM is just too much for it to work reliably. If it works in Flash then I can add some comments to the top of the example.

cmumford, 2020-11-09:
It's a little better if saving to flash. The UL/LL buttons seem to navigate focus (but seems to skip values, I wonder if there is a separate debouncing issue?). The UR button does nothing (I'm guessing that's expected). But, if I select anything (Large/small) buttons, and press the LR button then I get something like:

    id: r1.a val: L untouch focused

gfwilliams, 2020-11-10:
@allObjects do you have any thoughts about this?