espruino/EspruinoDocs #602: Module MTQQ

MaBecker, 2021-06-19:


MaBecker, 2021-06-19:
Testing MQTT connect and disconnect in `./espruino --telnet`via WebIDE and a local mosquitto broker. 

sample code 

```js

  var myBroker = "192.168.194.179", //"127.0.0.1",
      myTopic = "broker",
      myID = "abc";

  var mqtt = require("MQTT").create(myBroker);

  mqtt.on('connected', function() {
    console.log("Connected");
    mqtt.subscribe(myTopic+'/'+myID);
    mqtt.publish(myTopic, JSON.stringify({"info":"connected","id":myID}));
  });

  mqtt.on('publish', function (pub) {
    console.log("topic: "+pub.topic);
    console.log("message: "+pub.message);
    mqtt.publish(myTopic,JSON.stringify({"got":"a message"}));
  });

  mqtt.on('error',function() {
      console.log('error');
      setTimeout(mqtt.connect,1E3); 
  });

  mqtt.on('disconnected', function() {
     console.log("Disconnected");
     setTimeout(mqtt.connect,1E3);
  });

  setTimeout(()=>{mqtt.connect();},2E3);
```

When stopping the broker the module MQTT emits a disconnect, prints the error and then `Uncaught Error: Function "emit" not found!`

```js
Disconnected
ERROR: Connect failed (err 49)
Uncaught Error: Function "emit" not found!
 at line 2 col 9
!1,this.emit("error",c.message),this.emit("disconnected")}
        ^
in function called from system
> 
```

Similar error when stating and broker is offline 

```js
ERROR: Connect failed (err 61)
error
Disconnected
ERROR: Connect failed (err 49)
ERROR: Connect failed (err 49)
Uncaught Error: Function "emit" not found!
 at line 2 col 9
!1,this.emit("error",c.message),this.emit("disconnected")}
        ^
in function called from system
> 
```

Let me know when you need further details or test.

gfwilliams, 2021-06-21:
It's because you're doing `setTimeout(mqtt.connect,1E3);` - did you find that code anywhere, or was it something you came up with?

The issue is you're losing the value of `this`. JS figures out which object a method is called on based on *how* it is called. If you just pass `mqtt.connect` without calling it, it's like saying `fn = mqtt.connect;` and then just calling `fn()`.

If you do ` setTimeout(function() { mqtt.connect() },1E3);` it should work fine

closed 2021-06-21 by gfwilliams

MaBecker, 2021-06-21:
Wow - many thanks -  yes it works.

> did you find that code anywhere, 

no

> or was it something you came up with?

yes

gfwilliams, 2021-06-21:
Great! Yeah, it's one of those annoying JS quirks :)