espruino/EspruinoDocs #604: tutorials/Bangle.js Data Streaming.md

jeffreytchapman, 2021-06-24:
Hi,

Puck.js returns cannot connect when usitn the Try it button and after pasting into my own GitHub page.

Any suggestions?

Thanks in advance.

Jeff

gfwilliams, 2021-06-24:
Does any other Web Bluetooth code work for you?

Please could you open up the Chrome Developer Console and post what it says?

jeffreytchapman, 2021-06-24:
Wow, amazing response from the developer.  Below is the screen shot.  LED On! and LED Off! were inserted from the web tutorial and work fine.

Jeff

From: Gordon Williams
Sent: 24 June 2021 17:04
To: espruino/EspruinoDocs
Cc: jeffreytchapman; Author
Subject: Re: [espruino/EspruinoDocs] tutorials/Bangle.js Data Streaming.md(#604)

Does any other Web Bluetooth code work for you?
Please could you open up the Chrome Developer Console and post what it says?

gfwilliams, 2021-06-24:
Hi - sorry - the screenshot doesn't show up for me :(

Just tried here and the 'Try Now' link definitely works for me here though.

jeffreytchapman, 2021-06-29:
Hi Gordon.  Great product and great support.

I am an idiot twice.  First, I am trying Data Streaming on a Puck and not a Bangle.  Second, I changed Bangle to Puck in the code and it still does not work.  An alert box says ‘Couldn't connect!’  What changes should I make to the code for it to run on a Puck?  Screen shot below and attached.

Jeff

Jeff

From: Gordon Williams
Sent: 24 June 2021 19:12
To: espruino/EspruinoDocs
Cc: jeffreytchapman; Author
Subject: Re: [espruino/EspruinoDocs] tutorials/Bangle.js Data Streaming.md(#604)

Hi - sorry - the screenshot doesn't show up for me :(
Just tried here and the 'Try Now' link definitely works for me here though.

gfwilliams, 2021-06-29:
Hi - still no screenshots I'm afraid. Looks like you may have just replied to the email and you probably need to upload photos via GitHub.

If you're just using Puck.js, please could you try the 'Try Now' link on http://www.espruino.com/Web+Bluetooth#two-way-communications ? That will stream data from the Puck's light sensor in much the same way, but it should work as-is.

To use the Data Streaming example, you need to change the BANGLE_CODE contents so `Bangle.on('accel'` was `Puck.on('accel'`, and you also need to add `Puck.accelOn();` because the Puck's accelerometer isn't on by default

jeffreytchapman, 2021-06-29:
Hi Gordon,

All of the other Try Now links work and I need to use the accel.

I changed BANGLE_CODE contents so Bangle.on('accel' is Puck.on('accel', 

For the life of me, I cannot add  Puck.accelOn(); any where that works!  It either doese not connect, or the Puck.accelOn is not a function

 Any advice on how and where to place Puck.accelOn();

Thanks amazing support.

Jeff

From: Gordon Williams
Sent: 29 June 2021 11:50
To: espruino/EspruinoDocs
Cc: jeffreytchapman; Author
Subject: Re: [espruino/EspruinoDocs] tutorials/Bangle.js Data Streaming.md(#604)

Hi - still no screenshots I'm afraid. Looks like you may have just replied to the email and you probably need to upload photos via GitHub.
If you're just using Puck.js, please could you try the 'Try Now' link on http://www.espruino.com/Web+Bluetooth#two-way-communications ? That will stream data from the Puck's light sensor in much the same way, but it should work as-is.
To use the Data Streaming example, you need to change the BANGLE_CODE contents so Bangle.on('accel' was Puck.on('accel', and you also need to add Puck.accelOn(); because the Puck's accelerometer isn't on by default

gfwilliams, 2021-06-29:
Ok, just tested and it'd because the `Puck.accel` event is slightly different to the Bangle's. It reports `acc,gyro` rather than just `x,y,z`, and the units of the event are in the accelerometer's units (8192 = 1g) rather than just G.

Full code that is tested and works below:

```
<html>
 <head>
  <title>Bangle.js Accelerometer streaming</title>
 </head>
 <body>
<script src="https://www.puck-js.com/puck.js"></script>
<button id="btnConnect">Connect</button>
<p>X: <span class="bar"><span id="barX"></span></span></p>
<p>Y: <span class="bar"><span id="barY"></span></span></p>
<p>Z: <span class="bar"><span id="barZ"></span></span></p>
<script>
// Code to upload to Bangle.js
var BANGLE_CODE = `
Puck.on('accel',function(a) {
  var d = [
    "A",
    a.acc.x, a.acc.y, a.acc.z
    ];
  Bluetooth.println(d.join(","));
});
Puck.accelOn();
`;

// When we click the connect button...
var connection;
document.getElementById("btnConnect").addEventListener("click", function() {
  // disconnect if connected already
  if (connection) {
    connection.close();
    connection = undefined;
  }
  // Connect
  Puck.connect(function(c) {
    if (!c) {
      alert("Couldn't connect!");
      return;
    }
    connection = c;
    // Handle the data we get back, and call 'onLine'
    // whenever we get a line
    var buf = "";
    connection.on("data", function(d) {
      buf += d;
      var l = buf.split("\n");
      buf = l.pop();
      l.forEach(onLine);
    });
    // First, reset the Bangle
    connection.write("reset();\n", function() {
      // Wait for it to reset itself
      setTimeout(function() {
        // Now upload our code to it
        connection.write("\x03\x10if(1){"+BANGLE_CODE+"}\n",
          function() { console.log("Ready..."); });
      }, 1500);
    });
  });
});

// When we get a line of data, check it and if it's
// from the accelerometer, update it
function onLine(line) {
  console.log("RECEIVED:"+line);
  var d = line.split(",");
  if (d.length==4 && d[0]=="A") {
    // we have an accelerometer reading
    var accel = {
      x : parseInt(d[1]) * 100 / 8192,
      y : parseInt(d[2]) * 100 / 8192,
      z : parseInt(d[3]) * 100 / 8192,
    };    
    // Update bar positions
    setBarPos("barX", accel.x);
    setBarPos("barY", accel.y);
    setBarPos("barZ", accel.z);
  }
}
// Set the position of each bar
function setBarPos(id,d) {
  var s = document.getElementById(id).style;
  if (d>150) d=150;
  if (d<-150) d=-150;
  if (d>=0) {
    s.left="150px";
    s.width=d+"px";
  } else { // less than 0
    s.left=(150+d)+"px";
    s.width=(-d)+"px";
  }
}
</script>
<style>
/* Styles just to make the bars for X Y and Z look neat */
.bar {
  width : 500px;
  height: 24px;
  background-color : #D0D0D0;
  position:relative;
  display: inline-block;
}
.bar span {
  width : 1px;
  height: 20px;
  background-color : red;
  position:absolute;
  display: inline-block;
  left: 150px;
  top: 2px;
}
</style>
 </body>
</html>
```

closed 2021-06-29 by gfwilliams