espruino/EspruinoDocs #617: byte alignment in imagemagick

idavydov, 2021-11-17:
Hi,

`Graphics.md` has the following example of converting image to a base64 string.

```
# Create a 1 bit, 16x16 pixel black and white image called output.raw
convert myimage.png -resize 16x16\! -depth 1  gray:output.raw
```

Turns out, this only works as expected if image width is a multiply of 8. If not, the imagemagick pads bytes by zero bits. E.g., if you run the following code:

```
$ convert 9x12.gif -depth 1  gray:- | base64 
wYC+gEkASQB/AHcAfwBdAGMAfwC+gMGA
```

and then try to create an image object:
```
var img = {
  width : 9, height : 12, bpp : 1,
  transparent : 0,
  buffer : E.toArrayBuffer(atob("wYC+gEkASQB/AHcAfwBdAGMAfwC+gMGA"))
};

g.clear();
g.drawImage(img, 0, 0);
```

You will have a wrong image; the width now has to be changed to 16, then it will work as expected.

1) I would suggest adding this warning to the documentation.
2) I didn't find an easy way to perform this conversion from the command-line. I know about the [image convertor](https://www.espruino.com/Image+Converter), but I would prefer to do this from command-line, since it's multiple images.

gfwilliams, 2021-11-17:
Thanks! I'll get a note into the documentation now.

For the command-line, the code for the image converter is at https://github.com/espruino/EspruinoWebTools - I guess it probably wouldn't be too hard to create a little command-line node.js app that used that file in order to convert images.

Same goes for heatshrink.js as well. I don't have time right now but if someone felt like doing the CLI apps I'd be very happy to push EspruinoWebTools to npm so it was easy for others to use too

closed 2021-11-17 by gfwilliams

idavydov, 2021-11-21:
In case anyone stumbles upon this issue, I created a simple python [tool](https://github.com/idavydov/img2espruino) for conversion.

commit 8df0552d (2021-11-17): fix #617