espruino/EspruinoDocs #637: tutorials/Tiny Word Clock.md

schroepfer, 2022-03-18:
Hello,
your Tiny Word Clock works fine, but:
I save() the programm and want to set the time with the page: "https://www.espruino.com/Bluetooth+Time".
I paired and get "set time successfully".
But the 8x8 display goes dark and never goes on....

I can only set the time in the Web IDE with: setTime(Date.parse("2022-03-18T08:06:00") / 1000)

PS: I have a MDBT42Q with 2v12

regards
Juergen

gfwilliams, 2022-03-18:
Ok, thanks - the time setter webpage sends a Ctrl-C character first to clear the input line so I wonder whether that causes the setInterval loop to get broken out of. I'll take a look

schroepfer, 2022-03-24:
i tested it on different browsers, on a PC, MAC and iPhone. I installed an older Espruino version on my MDBT42q, installes your code fresh, ....nothing!

gfwilliams, 2022-03-28:
I'll give it a try when I get a moment. For now, you could just go to the IDE at https://www.espruino.com/ide/, choose to upload a blank file to 'RAM' and then click upload - it should automatically send the current time, then you can type `load()` on the left-hand side to reload the clock

gfwilliams, 2022-03-29:
I'm just pushing an update now which should allow this to work - there are now two switches. Either turn 'send break' off, or turn 'reload after send' on and you should be fine :)

closed 2022-03-29 by gfwilliams

schroepfer, 2022-03-29:
Hello Gordon,
thanks, but your solution doesn't work.
No matter which switch is on or off, the time is not set.
But the display is no longer dark! It shows the old time...

gfwilliams, 2022-03-30:
Did you definitely save the code you had to 'flash' in the IDE, or was it just in RAM?

If it was saved to RAM then the `reload` option would probably stop it working.

However, with `reload` *and* `send ctrl-c` unchecked, the time should automatically update *after one minute* - it's just not immediate

schroepfer, 2022-03-30:
Hello,
i did it, like you wrote in your example: https://www.espruino.com/Tiny+Word+Clock
:
_Software[⇠](https://www.espruino.com/Tiny+Word+Clock#software)
Just copy this to the right-hand side of the IDE and click Upload. When it's as you like it you can just type save() on the left-hand side to save the state permanently._

So i saved it to RAM and save().
Where did you wrote, that the program should upload to flash?

gfwilliams, 2022-03-30:
> So I saved it to RAM and save().

Ok, that's fine. `save()` will save from RAM to flash memory (more info at https://www.espruino.com/Saving if you're interested)

I just dug out my word clock, uploaded exactly the code from https://www.espruino.com/Tiny+Word+Clock and typed `save()` as it shows.

https://www.espruino.com/Bluetooth+Time seems to work fine for me now - with the default of both checkboxes checked. 

**However** because the clock is saved with `save()`, it's saved as an image with what's in RAM at the time it was saved, which includes the instructions for what to write to the screen (see the above link).

That's fine, but it means that when it 'reloads', the time shown is the time at which the code was originally written, **and you need to wait somewhere between 0-60 seconds before the `onMinute` interval fires and the time updates.

So are you certain that when you run https://www.espruino.com/Bluetooth+Time, the time doesn't update itself to the correct time after a minute as I said above?

schroepfer, 2022-03-31:
Now it seems to work!
I Think, i had a problem with the flash. 
I send an empty program to RAM, reset(), save() to flash and then the old clock program startet!?
From an empty RAM and empty flash?
I can't understand this....

Now i flashed the Firmware new, save and reset to RAM and FLASH and now it seems to work. Somehow.

gfwilliams, 2022-03-31:
> I send an empty program to RAM, reset(), save() to flash and then the old clock program startet!?

Maybe you had also written to flash one of the other ways mentioned in https://www.espruino.com/Saving ?

You can do `reset(1)` to clean out all saved code in whatever form it's in, so that could help.

Still, glad it's working now!

commit c8ec9899 (2022-03-29): Add option to reload after setting time - fixes #637