espruino/EspruinoDocs #655: Bangle2.js Firmware Source Code and Hardware Schematic

freetink3r, 2022-07-03:
I bought the Bangle2.js and a second one for my son so we could write our own firmware from the bare metal up, not just JavaScript apps. I also bought the nRF52 DK, because I saw the SWD pins are brought out. However, now that I have the hardware, I can't find the source code for the Bangle2.js firmware or the hardware schematic. When I saw the Bangle2.js advertised as the "Open Smart Watch", and I saw the SWD pins, I thought I'd be able to build our own thing based on the source code for the firmware. I can't find the low-level Bangle2.js firmware or documentation about how the hardware is connected. Am I missing something, or is the firmware and hardware schematic closed source? If so, this seems deceptive.

gfwilliams, 2022-07-04:
Hi - did you find everything you needed in the end?

freetink3r, 2022-07-04:
Yes! Thanks for following up! I was embarrassed by my original question, so removed the description.

I had some trouble connecting the Bangle.js2 to the nRF52 DK, but figured it out in the end. There's a bug in the current documentation, which references an image file nRF52DK.jpg, but the actual file is named nRF52DK.png. The image also doesn't show the short required between the `VTG` and `VDD` pins, which is necessary on the latest nRF52 DK boards. I created a pull request that addresses both of these issues: https://github.com/espruino/EspruinoDocs/pull/656

gfwilliams, 2022-07-05:
Great, thanks! Just merged - surprised about the VTG thing though - which board in particular are you using it with?

I took a quick look and have updated some links that I think were wrong, that may give you some more info: https://github.com/espruino/EspruinoDocs/commit/ae2a45f0dc0e1614ace507f5cc75844b4f0e0e28

If you do find anywhere we could link to things better, please let me know (or even better, PRs like your last one are great!).

closed 2022-07-05 by gfwilliams