espruino/EspruinoDocs #661: STM32F4DISCOVERY with different accelerometer

notEvil, 2022-08-13:
Hi,

the page

https://github.com/espruino/EspruinoDocs/blob/44f20731474274e693064ed087a25043e9203798/devices/LIS302DL.md

states

> This is the accelerometer that is on the STM32F4DISCOVERY board.

Unfortunately this is not always true, "newer" boards have a LIS3DSH whose SPI protocol is slightly different:

```js
R = 1 << 7;
W = 0;

function onInit() {
  SPI1.setup({mosi: A7, miso: A6, sck: A5});
  print('who am i:', SPI1.send([R | 0x0F, 0], E3));
  SPI1.send([W | 0x20, 0b01100111], E3);
}

var avrx=0.0, avry=0.0;
function getAcc() {
  var accx = SPI1.send([R | 0x28,0], E3)[1] | (SPI1.send([R | 0x29,0], E3)[1] << 8);
  var accy = SPI1.send([R | 0x2A,0], E3)[1] | (SPI1.send([R | 0x2B,0], E3)[1] << 8);
  if (accx>32767) accx-=65536;
  if (accy>32767) accy-=65536;
  avrx = 0.1*accx + 0.9*avrx;
  avry = 0.1*accy + 0.9*avry;
  digitalWrite(LED3, avrx > 8192);
  digitalWrite(LED2, avrx < -8192);
  digitalWrite(LED1, avry > 8192);
  digitalWrite(LED4, avry < -8192);
}
onInit();setInterval(getAcc, 10);
```

I know the board isn't officially supported, but I thought you might update this page if its just copy-paste :)

gfwilliams, 2022-08-15:
Thanks - looks like there's already a library that should work for the new parts? At least it seems to send roughly the same register configs: http://www.espruino.com/LIS3DH

I'll put a link in the docs though.

notEvil, 2022-08-15:
They are similar but not the same (sampling rates differ for instance). With the updated information this issue is sufficiently solved imo. Thanks!

(found http://forum.espruino.com/conversations/259300/ which mentioned the "new" acc 8 years ago ^^)

closed 2024-02-09 by gfwilliams