espruino/EspruinoDocs #688: info/Font Converter.md

Shimmy337, 2023-02-18:


Shimmy337, 2023-02-18:
Apologies, just tried it on Edge Browser and it works fine. It's just Brave browser that I had issues with.

closed 2023-02-18 by Shimmy337

gfwilliams, 2023-02-20:
Thanks - that's a strange one though - I wonder why it fails, as it should use the same rendering engine as Edge.

Even Firefox works ok which uses something different. The main text on the screenshot looks small though - do you have some custom zoom level set up? Maybe it would work if viewed at 100%? It still works ok for me with zoom in chrome, but maybe Brave does something different.

Shimmy337, 2023-02-25:
Hey again! Thanks for getting back to me.
Made sure my zoom was set to default and still not happy regardless of the zoom.
[image]

No fuss though, I'm fine to just use another browser to get the fonts.
Thanks for the help! :)