espruino/EspruinoDocs #695: ST7735 on ESP32 using hardware SPI

hagai-shatz, 2023-06-18:
I am trying the ST7735 display with an ESP32 board.

The suggested code works with software SPI (`var spi = new SPI();`) but the refresh speed is very slow.

When trying to use hardware SPI as below, the ESP32 is failing with an error due to the DMA buffer being too large (16bit * 128 pixels * 16 lines = 4096 bytes, ESP32 DMA max buffer size is 4095).

This can be fixed by reducing the [number of lines that are sent in each iteration](https://github.com/espruino/EspruinoDocs/blob/452bd18a5ee5155f0d63402617fe1c39f831d6a1/devices/ST7735.js#L103) from 16 to 8:
```js
var lines = 8; // size of buffer to use for un-paletting
```

The result is a much faster refresh of the screen due to the hardware SPI speed.

Code to use ST7735 with hardware SPI:
```JS
SPI1.setup({mosi:D19 /* sda */, sck:D18 /* scl */,baud:20000000});
var g = require("ST7735").connect({
  palette:colorPalette,
  spi:SPI1,
  dc:D21,
  cs:D15,
  rst:D2,
  // height : 160 // optional, default=128
  // padx : 2 // optional, default=0
  // pady : 3 // optional, default=0
}, function() {
  g.clear();
  g.setColor(3);
  g.drawString("Hello",0,0);
  g.setFontVector(20);
  g.setColor(1);
  g.drawString("Espruino",0,10);
  g.setColor(2);
  g.drawString("Espruino",0,40);
  g.flip(); //<--- Send to the display
});
```

hagai-shatz, 2023-06-18:
Created https://github.com/espruino/EspruinoDocs/pull/696 for this

gfwilliams, 2023-06-19:
Ok, thanks! I think ideally this is something that would be nice to fix inside Espruino - there seems no reason it couldn't just batch up writes of 4095 bytes automatically, but if that PR fixes it hopefully the performance impact on other devices won't be that big

gfwilliams, 2023-06-19:
Closed with https://github.com/espruino/EspruinoDocs/pull/696

closed 2023-06-19 by gfwilliams