espruino/EspruinoDocs #702: Add "Bangle.js Battery Load/Intensity notes"

thyttan, 2023-08-27:
On the forum some knowledge of what contributes to faster battery drain has surfaced. I think it could be helpful to compile these in a page maybe similar to the [Espruino Performance Notes](http://www.espruino.com/Performance).

If I get around to compiling such a page I'll search the forum first and also start a forum thread asking for the members input.

Please comment below with e.g. tips to add to the page.

thyttan, 2023-08-27:
There is a section in the wiki, maybe just add to that if I find something that's missing.

closed 2023-08-27 by thyttan

gfwilliams, 2023-08-29:
Good idea - there is a mention of things that draw power on http://www.espruino.com/Bangle.js2#power-consumption but maybe we could also point to the wiki from there, or elaborate a little?