espruino/EspruinoDocs #713: Add Ungoogled Chromium and Bromite to docs as a FOSS alternative

Darin755, 2024-01-12:
I have been using Bromite and Ungoogled Chromium to access the banglejs apps site. These work well and they are similar to chrome and are chromium based. These browsers are much more FOSS than Chrome or Edge and do not include the scary telemetry. I don't think it is necessary to add a bunch or extra content for these browsers but it would be nice to mention that they work. I am more than happy to create a PR just let me know what you think.

gfwilliams, 2024-01-15:
Yes, I'm open to mentioing this in the docs - a PR would be great. Where were you thinking of mentioning it?

It seems like https://www.espruino.com/Quick+Start+BLE#with-web-bluetooth might be a reasonable place since https://banglejs.com/start2 links there as well

Darin755, 2024-01-15:
I'll spend some time looking into what edits need to be made. It should be pretty simple to add a note about private alternatives.

closed 2024-02-09 by gfwilliams