espruino/EspruinoDocs #723: New tutorial: How to rapidly build and test Espruino for banglejs2

atjn, 2024-03-23:
This is a continuation of https://github.com/espruino/Espruino/pull/2476#issuecomment-2009488313

I want to quickly change some code in Espruino, then compile it and flash it to my watch or the web ide emulator so that I can test my changes. That process should be very fast and very safe - no risk of breaking my watch just because I want to test a small change.

I am not able to find any documentation that can help me with that, so I want to write a tutorial specifically for this use case. I will probably need your help in finding a solution that works at all.

It should cover:

- [x] How to compile espruino - you mentioned the github action, that seems like the best solution
- [ ] How to upload it easily to the banglejs2. Preferrably you can do this directly from the PC you are developing on via bluetooth.
  - Using the App Loader's `fwupdate`? - fails for me, the updater looks for a `manifest` file in my zip file and breaks when it doesn't exist.
  - Using the web IDE? - fails for me, no error message
  - (I also tried flashing via the recommended android app, this also fails with no error message)
- [ ] How to make sure the upload cannot brick the device
- [ ] Alternatively, how to build the Web IDE with an emulator that uses the custom espruino build

I think this would be a very useful guide, what do you think?

gfwilliams, 2024-03-25:
Hi,

yes, this would be a good guide. I think https://github.com/espruino/Espruino/blob/master/README_Building.md#super-easy-github-method is probably what you need though.

> To perform a DFU update on an nRF52 device, you need to unzip the BOARDNAME.zip file and then the zip file inside that called espruino_2vXX.YY_boardname.zip is the one you need to use for the firmware updates. On MacOS it may not be immediately apparent that the thing in BOARDNAME.zip named espruino_2vXX.YY_boardname is another zip file, and not a folder.

So linking to that from the tutorial might be good - I think that answers your second question.

> How to make sure the upload cannot brick the device

Well, to some extent you can't. But as long as you're not messing with the bootloader you should always be able to recover by flashing this way: https://www.espruino.com/Firmware+Update#nrf52

> how to build the Web IDE with an emulator that uses the custom espruino build

https://github.com/espruino/EspruinoWebIDE/blob/master/extras/create_emulator.sh

But the emulator is really got a great test for whether things will work on a real Bangle

atjn, 2024-03-27:
Thanks for the links, that helped a lot! I am waiting to see if the two open PR's can be merged, then I should be ready with the tutorial :)

PR #725 (2024-04-02): New tutorial: Testing Espruino on Bangle.js 2
Closes #723
closed 2024-04-02 by gfwilliams