espruino/EspruinoDocs #726: tutorials/Tiny Word Clock.md

brendonhatcher, 2024-04-05:
https://www.espruino.com/Tiny+Word+Clock
My clock is no longer working correctly. Individual letters light up, but they are the wrong letters.
I have upgraded the firmware to 2v21.
If I open the code in the Web IDE...

Line 17: "a":[00],

Error: Don't use extra leading zeros '00'

However, the 00 is the X and Y coordinates of the letter A on the grid, and the whole script uses 2 digits per character, so this isn't a simple tweak to fix (certainly beyond my skills).

gfwilliams, 2024-04-09:
Hi - that 'error' in the IDE is just a warning about code style - I don't believe that is the reason for your problems...

I have just installed 2v21 on my original Tiny Word Clock from the https://www.espruino.com/Tiny+Word+Clock tutorial, re-uploaded the exact code and it works perfectly with all the same LEDs as before.

Is it possible that maybe the first JS code you used, from before you updated the firmware, you'd used different pin numbers? You might have soldered your board on one pin out, or maybe the board is soldered to the LED matrix 180 degrees out from the way I did it, in which case the `ANODES` and `CATHODES` variables would need changing?

brendonhatcher, 2024-04-09:
Thanks for replying.  I don't think I did anything unusual in my original setup (years ago).  Is there any way to output the word that it is trying to display to the Web IDE?

gfwilliams, 2024-04-10:
Sure, yes - if you go to the function `function outputWords(wordList) {` and add `console.log(wordList)` after that line then it should tell you

brendonhatcher, 2024-04-10:
That was very helpful.
My output is rotated 180 degrees.
However, it used to work!

Can you suggest how I change the code?

Regards
Brendon

gfwilliams, 2024-04-10:
Well, I think you probably must have had some code change in there! I vaguely remember someone asking a question about rotation a while back so it might have been you?

After `var g = Graphics.createArrayBuffer(8,8,1);` just add `g.setRotation(2);` and I bet that'll sort it for you!

brendonhatcher, 2024-04-10:
It worked!
Thanks

gfwilliams, 2024-04-10:
Great! I'll put a comment in the tutorial code about rotation

closed 2024-04-10 by gfwilliams