espruino/EspruinoDocs #727: Mouldking module not working, possibly wrong device address?

sebastian-ruiz, 2024-04-09:
I am using an esp-WROOM-32 dev kit. I am trying to use [the mouldking.js module](http://www.espruino.com/modules/mouldking.js) but I am having trouble with it. 

I have the [Mould King 4.0 Powered Module](https://www.amazon.de/-/en/dp/B09KY8LL3Y).

When I run the code on [http://www.espruino.com/LEGO+Power+Functions+Clone](http://www.espruino.com/LEGO+Power+Functions+Clone) nothing happens. When I use my iphone to connect to the mouldking, the frequency of the flashing light changes on the mouldking powered module. This doesn't happen when I run the code.

I tried to scan for Mould King packets from the app with UUID 65280 / 0xFF00, as written in the notes on that page, but I don't get any packets with that manufacturerData.

When I open the iphone app and connect, it is sending packets like this:
```
BluetoothDevice: {
  "id": "7b:fd:01:8b:6e:5e random",
  "rssi": -53,
  "data": new Uint8Array([2, 1, 26, 27, 3, 249, 8, 73, 34, 71, 186, 196, 188, 19, 10, 219, 130, 101, 177, 189, 11, 37, 9, 125, 210, 18, 19, 20, 21, 22, 23]).buffer,
  "services": [
    "08f9",
    "2249",
    "ba47",
    "bcc4",
    "0a13",
    "82db",
    "b165",
    "0bbd",
    "0925",
    "d27d",
    "1312",
    "1514",
    "1716"
   ]
 }
```

I don't see these same numbers in the [mouldking.js](http://www.espruino.com/modules/mouldking.js). Maybe mine is a different version?

I don't see the hello message packet that the iphone is sending.

Thank you very much for your help and for your time. Any help would be greatly appreciated.

sebastian-ruiz, 2024-04-10:
Thanks for your reply. 

I used the [Mould King iOS app](https://apps.apple.com/us/app/%E5%AE%87%E6%98%9F%E7%A7%91%E6%8A%80/id1457417598?platform=iphone) and both the "4.0 Module" and "4.0 Module (old)" work for mine:

[image]

Are either of these devices ones that you looked at in the disassembled android app?

I have an nRF52840 BLE sniffer, so I will try if I can sniff useful info using Wireshark, but I'm not experienced with this.

gfwilliams, 2024-04-11:
Well, you can see in the picture on https://www.espruino.com/LEGO+Power+Functions+Clone the module I tested with - it's the M-0006 kit - it sure looks like the "4.0 Module (old)" is the same one.

Looking again at the disassembled code, the only advertising it ever does is with manufacturerID as 65280 (not services) so it looks like it's just not compatible with the same protocol. In my disassembled version the advertising is set in the `BluetoothAdvertiseThread` class so maybe if you downloaded the Android app that matches your device, then disassembled it (there are online disassemblers) you might be lucky and find a `BluetoothAdvertiseThread` class that matches and shows how data is encoded.

Looking at the 'data' in what you posted, there's:

```
2, 1,  // header, standard bluetooth advertising flags
26,  // flags
27, 3, // header, 16 bit service list
249, 8,  // 16 bit services...
73, 34,  // ...
71, 186, 
196, 188, 
19, 10, 
219, 130, 
101, 177, 
189, 11, 
37, 9, 
125, 210, 
18, 19, 
20, 21, 
22, 23
```

So it looks like they're just encoding the data as a list of service UUIDs (which is totally not supposed to be allowed in the BLE spec). It's *possible* they're doing that because you're on iOS and maybe iOS doesn't let you advertise manufacturer data? So it'd be interesting if you have an Android device to see what that advertises instead.

You shouldn't really need the BLE sniffer (although you could use it) - assuming there is no actual BLE connection (it's all done with advertising) you can just use Espruino to keep scanning for advertising packets and dumping as you have been - and see if there are any noticeable patterns when you try and do different things in the app.

If you do something, then return to doing nothing and the advertising data stays the same as it was the last time, there's a good chance it'll be easy to decode (if only by replaying the patterns).

But judging by how my one worked, they went well out of their way to obfuscate the data - so it may be it's not trivial to decode (unless they used the exact same algorithm and just dumped the data in as service data rather than manufacturer data)