espruino/EspruinoDocs #737: Bangle 1 firmware update instructions missing

timp, 2024-09-28:
https://www.espruino.com/Firmware+Update#nrf52 does not give guidance on updating the original Bangle 1 firmware.

timp, 2024-09-28:
Web IDE Flasher worked, but the modal does  not tell you it can be closed when the progress bar is complete.

gfwilliams, 2024-09-30:
There do seem to be instructions under `Entering DFU mode` along with the other boards?

[image]

Is your issue that Bangle.js 1 isn't specifically called out at the top? The reason Bangle.js 2 is called out is because usually we want users to update the firmware a different way on that.

> but the modal does not tell you it can be closed when the progress bar is complete.

I've just checked and the modal should be closed automatically, but I know on Bangle.js 1 I have noticed times where the update gets to 99% and doesn't signal completion (even though it's fine), so that could be it. It shouldn't happen all the time though.

timp, 2024-09-30:
Thank you so much for your reply. 
I have both my bangle 1 and bangle 2 updated now.

closed 2024-09-30 by timp