espruino/EspruinoDocs #747: Question: detection of fall and impact?

jbrepogmailcom, 2025-04-03:
Hello, are there any plans to send alerts when fall or impact are detected?

Accelerators can usually be configured to run in low power mode (microAmps or nanoAmps) and monitor if acceleration reaches a threshold. Then generate interrupt. This can be utilized by the watch to emit BLE event.

Are there any works on that? Thanks, Jan

gfwilliams, 2025-04-03:
The Bangle's accelerometer won't detect low-g automatically (but it can detect movement - see the datasheet at 
 https://www.espruino.com/datasheets/KX022-1020.pdf if you're interested)

But it's trivial to just detect it in code yourself, for example:

```
var fallCount = 0;
Bangle.on('accel', function(a) { 
  if (a.mag < 0.4) 
    fallCount++;
  else
    fallCount = 0;
  if (fallCount==2)
    print("Fall Detected");
});
```

The power usage from that isn't that high. Although the accelerometer runs at 12.5Hz by default - so perhaps for more realistic detection you want a higher data rate (which you can configure pretty easily as well if needed).

I don't want to build this in though as generally when someone asks for this it's because they want to detect when an elderly person falls. And I don't want to be held responsible if they fall and it isn't detected - if you're willing to take that risk then as you see above it's pretty easy to implement your own algorithm though!

If you've got other questions please can you ask them in the forum though? http://forum.espruino.com/

closed 2025-04-03 by gfwilliams

jbrepogmailcom, 2025-04-03:
Thanks for the answer. You are right that detecting 0g in KX022-1020 is unclear in documentation. Detecting crash or impact could be possible, I have in mind accelerations like 10 or 20g, but that depends on range to which the accelerator is set. Most likely the watch will be set to most sensitive range like 2g. Is it possible to change range to more g?

gfwilliams, 2025-04-03:
I think it's 4g by default. You can reconfigure it but the Bangle.js won't be aware so all the other readings it gives will be different.